dvarrel/XCTRACK_vario #1: Why won't this work with Adafruit Feather M0?

glydrfreak, 2020-07-12:
I've successfully tested this with Arduino nano.

I'm trying to get it to work with an Adafruit Feather M0 (with native USB), but no luck.

Both boards are running the same code, but I cannot establish a connection when using the Adafruit Feather M0. Any thoughts?

vspeedvario@gmail.com

dvarrel, 2020-07-13:
Hello, i think USB driver is different, and xctrack does not support it

glydrfreak, 2020-07-13:
That sure would be nice to have XCtrack support it. Thanks for the info!

The reason I asked is because I want to send pressure data as frequent as possible. I can get about 55 samples per second with the arduino nano, and somewhere on the order of 85 samples per second with an M0 processor. Is there a reason you limit the data transfer to three times per second in the code?

dvarrel, 2020-07-13:
You re right, it s better to send more than 3 Times a second

moby74, 2020-07-22:
Hello,

         I'm beginer  and when i try to upload the code, with bluino (on android phone), the error message indicating that the file MS5611.h is not found appears.

Could you help me?

Thank you!
Laurent

dvarrel, 2020-07-22:
Hello, you have to download ms5611 library , see m'y github readme

dvarrel, 2020-07-22:
> That sure would be nice to have XCtrack support it. Thanks for the info!
>
> The reason I asked is because I want to send pressure data as frequent as possible. I can get about 55 samples per second [...]

hello, with max resolution (0.012mbar with OSR 4096) it takes about 15ms, so we can't read more than 65Hz. 
what frequency should be ok, with paragliding ? 10Hz should be enough ?

closed 2020-07-22 by dvarrel

moby74, 2020-07-23:
Thank you Dvarrel. I loaded the library MS5611 and it resolved the problem.
For the ARV/WDT, I don't know if i have the right thing because during upload, i have a eeror message for the lines 63/67: _"cast from pointer to smaller type 'unint16' (aka 'unsigned short') loses information"_

Thank you for your help.
Laurent

dvarrel, 2020-07-23:
yes, perhaps you have to see how manage the watchdog configuration with M0,
or just comment the lines concerned
---
Garanti sans virus, je travaille sous Linux !