dvarrel/XCTRACK_vario #3: China clone Arduino Nano with CH340G chip connect to XCTrack

abinlee, 2020-09-14:
Hello,
Have you built vario by China clone Arduino Nano with CH340G chip and connect to XCTrack without problem?
I make up a vario with CH340G nano and found XCTrack 0.9.4.4 will always output CH340 connection initiate error.
https://gitlab.com/xcontest-public/xctrack-public/-/issues/523
Or if you successfully connect vario with China clone Arduino Nano with CH340G chip to XCTrack, may I know which XCTrack version you installed on your phone?

Regards,
Abin

dvarrel, 2020-09-14:
hello,
i never add errors, it works with arduino nano with ftdi chip and since
xctrack 0.9.3, ch340g chip. I've tested with ftdi and ch340g, and now i
only use ch340g

my actual version is 0.9.4.5-beta. I've got friends which use the non-beta
version
---
Garanti sans virus, je travaille sous Linux !

abinlee, 2020-09-15:
Hello,
After update XCTrack to 0.9.4.5-beta, it’s also not work on my phone.
I can see vario output by other USB serial app on my phone but XCTrack.
Maybe problem is from XCTrack.
I’ll close this issue and keep to debug with XCTrack developer over https://gitlab.com/xcontest-public/xctrack-public/-/issues/523.

Thanks,
Abin

closed 2020-09-15 by abinlee