dvarrel/XCTRACK_vario #4: Which accuracy do you have?

FeelUsM, 2020-09-29:
I use BMP180 sensor, and I have [this sample](https://yadi.sk/i/NqZ4Edri2J9fFQ) when sensor is rest.
I wont to compare BMP180 and ms5611 sensors.
(settings / Connections and sensors / Calibrate)

dvarrel, 2020-09-30:
hello,
i use ms5611 instead of bmp180, because it's much more accurate. when i
have time i'll give you a try with xctrack

---
Garanti sans virus, je travaille sous Linux !

dvarrel, 2020-09-30:
[image]

dvarrel, 2020-09-30:
[image]

dvarrel, 2020-09-30:
LPS22HB is also more accurate than ms5611. I've never test it. it's in swiss varios .

closed 2021-01-22 by dvarrel