dvarrel/XCTRACK_vario #5: Which versions of libraries do you use

jschustr, 2021-03-06:
Hi, first of all, thanks for your effort in this project. 
I am trying to make alive vario , but I have a problems with compiling your code. 
I ordered this Nano:
https://www.aliexpress.com/item/33013146682.html?spm=a2g0s.9042311.0.0.7a3a4c4dmAtUji

And then I am trying to compile your code I got these errors. I think there is a connection between errors and libraries version. Libraries wire.h and ms5611.h . Can you specify please, with wich versions of libraries should I try to compile your code?

Thanks a lot
Jan

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Honza\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Honza\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Honza\Documents\Arduino\libraries -fqbn=arduino:avr:nano:cpu=atmega328old -vid-pid=1A86_7523 -ide-version=10813 -build-path C:\Users\Honza\AppData\Local\Temp\arduino_build_480302 -warnings=none -build-cache C:\Users\Honza\AppData\Local\Temp\arduino_cache_907924 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Honza\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\Honza\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avrdude.path=C:\Users\Honza\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\Honza\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Honza\Documents\Arduino\vario original\XCTRACK_vario-master\test_MS5611\test_MS5611.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Honza\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Honza\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Honza\Documents\Arduino\libraries -fqbn=arduino:avr:nano:cpu=atmega328old -vid-pid=1A86_7523 -ide-version=10813 -build-path C:\Users\Honza\AppData\Local\Temp\arduino_build_480302 -warnings=none -build-cache C:\Users\Honza\AppData\Local\Temp\arduino_cache_907924 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Honza\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\Honza\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avrdude.path=C:\Users\Honza\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\Honza\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Honza\Documents\Arduino\vario original\XCTRACK_vario-master\test_MS5611\test_MS5611.ino
Using board 'nano' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "C:\\Users\\Honza\\AppData\\Local\\Temp\\arduino_build_480302\\sketch\\test_MS5611.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for Wire.h: [Wire@1.0]
ResolveLibrary(Wire.h)
  -> candidates: [Wire@1.0]
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\Wire\\src" "C:\\Users\\Honza\\AppData\\Local\\Temp\\arduino_build_480302\\sketch\\test_MS5611.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for MS5611.h: [Arduino-MS5611-master]
ResolveLibrary(MS5611.h)
  -> candidates: [Arduino-MS5611-master]
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\Wire\\src" "-IC:\\Users\\Honza\\Documents\\Arduino\\libraries\\Arduino-MS5611-master" "C:\\Users\\Honza\\AppData\\Local\\Temp\\arduino_build_480302\\sketch\\test_MS5611.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\Wire\\src" "-IC:\\Users\\Honza\\Documents\\Arduino\\libraries\\Arduino-MS5611-master" "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\Wire\\src\\Wire.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\Wire\\src" "-IC:\\Users\\Honza\\Documents\\Arduino\\libraries\\Arduino-MS5611-master" "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\Wire\\src\\utility\\twi.c" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\Wire\\src" "-IC:\\Users\\Honza\\Documents\\Arduino\\libraries\\Arduino-MS5611-master" "C:\\Users\\Honza\\Documents\\Arduino\\libraries\\Arduino-MS5611-master\\MS5611.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Generating function prototypes...
[... cut ...]
Could not find boards.txt in C:\Users\Honza\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.3.2. Is it pre-1.5?
Could not find boards.txt in C:\Users\Honza\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.3.2. Is it pre-1.5?
Could not find boards.txt in C:\Users\Honza\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.3.2. Is it pre-1.5?
Could not find boards.txt in C:\Users\Honza\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.3.2. Is it pre-1.5?
Could not find boards.txt in C:\Users\Honza\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.3.2. Is it pre-1.5?

jschustr, 2021-03-06:
ok, I found the problem thanks 
https://forum.arduino.cc/index.php?topic=719075.0
I used MS5611 library but I should use ms5xxx library. Now it works , thanks

closed 2021-03-06 by jschustr

dvarrel, 2021-04-10:
yes MS5XXX library by Roman Schmitz is the good one