dvarrel/XCTRACK_vario #6: error when verifying code

JollyRogers22, 2021-04-10:
When I try to verify/compile the code, I get this error msg:

Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Placa:"Arduino Nano, ATmega328P"

C:\Users\jpdes\Documents\Arduino\teste\teste.ino:19:15: warning: invalid conversion from 'TwoWire*' to 'uint8_t {aka unsigned char}' [-fpermissive]

teste:56:14: error: 'class MS5611' has no member named 'connect'; did you mean 'convert'?

   if (sensor.connect()>0) {

              ^~~~~~~

              convert

teste:67:10: error: 'class MS5611' has no member named 'ReadProm'; did you mean 'readProm'?

   sensor.ReadProm(); //takes about 3ms

          ^~~~~~~~

          readProm

C:\Users\jpdes\Documents\Arduino\teste\teste.ino: In function 'void loop()':

teste:77:10: error: 'class MS5611' has no member named 'Readout'

   sensor.Readout(); // with OSR4096 takes about 10ms

          ^~~~~~~

teste:78:30: error: 'class MS5611' has no member named 'GetPres'

   uint32_t Pressure = sensor.GetPres();

                              ^~~~~~~

teste:86:27: error: 'class MS5611' has no member named 'GetTemp'

     uint8_t Temp = sensor.GetTemp()/100;

                           ^~~~~~~

exit status 1

'class MS5611' has no member named 'connect'; did you mean 'convert'?

dvarrel, 2021-04-10:
hello,
do you have MS5XXX library by Roman Schmitz installed ?

JollyRogers22, 2021-04-14:
I was using MS5611 library... :-/

closed 2021-04-14 by JollyRogers22