dvarrel/XCTRACK_vario #7: More comprehensive documentation

carlosvbhenriques, 2021-11-09:
Hello,

I believe there could be more documentation explaining the steps, since it's rather simple when you know it.

I have done some changes locally but can't push and open a pull request. 

[Here are the changes](https://imgur.com/a/K7ZgDQM)

Thank you.

dvarrel, 2021-11-10:
hello, thank you for the job. if you can't open a ull request, can you attach file ?

carlosvbhenriques, 2021-11-10:
[You can add contributors here](https://i.imgur.com/Qk866Qa.png)

It might be easier to just do the pull request because you'll be able to see the changes more easily, but I'll send you the files I've changed and added: https://we.tl/t-E54BBpfE2F

* The change on the code itself was just updating some comments and settings usb on by default (disabling bluetooth) and saying the same the other way around. Just now I've realized you may want both on (fixing via git would be easier, if you want I can do it).
* Updated the readme
* Added an image of an issue I've ran into

dvarrel, 2021-11-10:
i've add you as contributor, thanks

carlosvbhenriques, 2021-11-10:
Opened the pull request and noticed that by default I can merge to master; You should disable that so that only you can merge into master. If there is any other changed needed let me know or do it and then merge.

closed 2021-11-10 by carlosvbhenriques

dvarrel, 2021-11-10:
"You should disable that so that only you" : i don't find how do it

dvarrel, 2021-11-10:
temperature can be added to xctrack in future version. LK8000 is the communication protocol between xctrack and arduino