dvarrel/XCTRACK_vario #10: STL files

lenkman, 2022-06-01:
How is it possible to get the STL files for printing without an onshape account?

dvarrel, 2022-06-01:
Yes i can put them in github

lenkman, 2022-06-01:
That would be great :)

I am gonna trying out this project, because it looks very interesting.

Good job 👍

dvarrel, 2022-06-01:
there are in stl folder

lenkman, 2022-06-01:
You're really fast :-)

Thanks for the STL files. When you have the time it would be great to have
the USB C type with and w/o Bluetooth as well.

I just build the first vario with a arduino nano.

I had the following problem when setting up the device in XCtrack:

Jumping values when having connected an GXAirCom via Bluetooth in parallel.

Solution to fix this was deactivate the vario output in the GxAirCom
settings.

Now it's working fine.

Best regards
Philip

-