dvarrel/XCTRACK_vario #11: no proper connection to xctrack

HCAHofm, 2022-09-10:
Hi first of all thanks for this nice project.
Unfortunately I cannot make it work. uploading the code on the Arduino went well. But xctrack cannot talk to the sensor. What am I doing wrong?
Below further information.

Xctrack version 0.9.8.3
CH340 nano (China clone)

Log output
2022-09-10 16:58:45.513 DEBUG [Info] adding loc: SensorsLocation{isReal=true, isValid=true, time=1662829127000, rolloverOffset=RolloverOffset{builtin=0, configured=0}, coord=LonLat{lon=6.7769006166666665, lat=51.24179838333333}, altGps=14.7, speedGps=0.0, bearingGps=0.0, headingCompass=57.98251160316778, airspeedSensor=null, pressure=null, speedComputed=0.03495452712290623, bearingComputed=17.38103624011674, headingComputed=0.0, airspeedComputed=0.0}
2022-09-10 16:58:46.513 DEBUG [Info] adding loc: SensorsLocation{isReal=true, isValid=true, time=1662829128000, rolloverOffset=RolloverOffset{builtin=0, configured=0}, coord=LonLat{lon=6.7769091, lat=51.241803266666665}, altGps=14.9, speedGps=0.0, bearingGps=0.0, headingCompass=56.99076439266861, airspeedSensor=null, pressure=null, speedComputed=0.802240554314016, bearingComputed=47.40149676688947, headingComputed=0.0, airspeedComputed=0.0}
2022-09-10 16:58:46.644 DEBUG [USBConnection] USB: line:>XCTRACK VARIO Sep  4 2022<
2022-09-10 16:58:46.645 DEBUG [lineNMEA] ExternalSensor: invalid nmea checksum: XCTRACK VARIO Sep  4 2022
2022-09-10 16:58:46.645 DEBUG [USBConnection] USB: line:>checking MS5611 sensor...<
2022-09-10 16:58:46.646 DEBUG [lineNMEA] ExternalSensor: invalid nmea checksum: checking MS5611 sensor...
2022-09-10 16:58:46.646 DEBUG [USBConnection] USB: line:>Error connecting MS5611...<
2022-09-10 16:58:46.648 DEBUG [lineNMEA] ExternalSensor: invalid nmea checksum: Error connecting MS5611...
2022-09-10 16:58:46.650 DEBUG [USBConnection] USB: line:>Reset in 1s...<
2022-09-10 16:58:46.651 DEBUG [lineNMEA] ExternalSensor: invalid nmea checksum: Reset in 1s...

dvarrel, 2022-09-11:
hello,
i see in your log that your card can't connect to ms5611. so it's not a link problem with xctrack because the program doesn't go after setup() and reset. check connection between ms5611 and your nano card

HCAHofm, 2022-09-12:
thanks. i will check and reply afterwards

HCAHofm, 2022-12-10:
Hi,

sorry for taking so long time to respond.
It was as you said. It was a connection problem.
Thanks again for your work!

closed 2022-12-10 by HCAHofm