dvarrel/XCTRACK_vario #12: Bluetooth Connection via HC-06 to XCTrack fails

PreisTo, 2022-10-02:
Hey,
first of all I want to thank you for this awesome project!

I did manage to make it work on a CH340 China clone, however only using USB not Bluetooth.
My Bluetooth module is a HC-06, however XCTrack returns following error:
[2022-10-02-14.log](https://github.com/dvarrel/XCTRACK_vario/files/9693227/2022-10-02-14.log)

`2022-10-02 18:30:28.025 ERROR [org.xcontest.XCTrack.l:a()12] Bluetooth exception while connecting
	EXCEPTION: java.io.IOException: read failed, socket might closed or timeout, read ret: -1
java.io.IOException: read failed, socket might closed or timeout, read ret: -1
	at android.bluetooth.BluetoothSocket.readAll(BluetoothSocket.java:758)
	at android.bluetooth.BluetoothSocket.readInt(BluetoothSocket.java:772)
	at android.bluetooth.BluetoothSocket.connect(BluetoothSocket.java:404)
	at org.xcontest.XCTrack.l.a(BluetoothSensorWorker.java:8)
	at org.xcontest.XCTrack.l.run(BluetoothSensorWorker.java:4)
	at java.lang.Thread.run(Thread.java:923)`

Because I'm pretty new to this matter I don't have any clue if its a matter on XCTrack's or the programs side. I attached the full log-file (at line 551 I switched to USB and afterwards the program runs as expected).

Thanks for your help!

dvarrel, 2022-10-02:
Hello, 
Test  bluetooth sensor with  app "serial bluetooth terminal"

PreisTo, 2022-10-03:
Thanks for the tip:
With my Oneplus 7T i get the same error message as in XTrack (read failed), however on a LG G7 the app prints out the error "writecharacteristics not writeable".
I have the feeling I got a faulty module, communication via "AT" aren't possible as well. Ordered another module and will test it.

dvarrel, 2022-10-05:
i test with a arduino nano and HC-07 module ( withour sensor )
```07:24:22.750 Connecting to DAM3 ...
07:24:23.607 Connected
07:24:24.555 XCTRACK VARIO V1.0
07:24:24.567 compiled Oct  5 2022
07:24:24.587 checking MS5611 sensor...
07:24:24.610 Error connecting MS5611...
07:24:24.644 Reset in 1s...
07:24:25.807 XCTRACK VARIO V1.0
07:24:25.809 compiled Oct  5 2022
07:24:25.811 checking MS5611 sensor...
07:24:25.836 Error connecting MS5611...
07:24:25.870 Reset in 1s...```

dvarrel, 2022-10-05:
and with XCTACK 0.9.8.3 :
```
2022-10-05 05:33:55.578 INFO [mainActivity] Preferences changed (Testing.Log), reloading.
2022-10-05 05:33:56.001 DEBUG [org.xcontest.XCTrack.l:b()2] BT: line:>XCTRACK VARIO V1.0<
2022-10-05 05:33:56.003 DEBUG [lineNMEA] ExternalSensor: invalid nmea checksum: XCTRACK VARIO V1.0
2022-10-05 05:33:56.033 DEBUG [org.xcontest.XCTrack.l:b()2] BT: line:>compiled Oct  5 2022<
2022-10-05 05:33:56.034 DEBUG [lineNMEA] ExternalSensor: invalid nmea checksum: compiled Oct  5 2022
2022-10-05 05:33:56.073 DEBUG [org.xcontest.XCTrack.l:b()2] BT: line:>checking MS5611 sensor...<
2022-10-05 05:33:56.075 DEBUG [lineNMEA] ExternalSensor: invalid nmea checksum: checking MS5611 sensor...
2022-10-05 05:33:56.084 DEBUG [org.xcontest.XCTrack.l:b()2] BT: line:>Error connecting MS5611...<
2022-10-05 05:33:56.085 DEBUG [lineNMEA] ExternalSensor: invalid nmea checksum: Error connecting MS5611...
2022-10-05 05:33:56.102 DEBUG [org.xcontest.XCTrack.l:b()2] BT: line:>Reset in 1s...<
```

closed 2022-10-05 by dvarrel