dvarrel/XCTRACK_vario #13: Usb connection fail

liegestuhl, 2022-10-03:
Hey,
thanks for this nice project.

I have a connection problem with the usb connection. The arduino clone (HIMALAYA basic 5V 16MHz ATMEGA328P Board Compatible with Arduino Nano CH340 mini-USB UART IC) seems to work quite well, because with the app USB Terminal I can see the data string as expected. But with XCTrack 0.9.8.3 there is nothing, and the program returns the following error as seen in the log file.
```
2022-10-03 07:41:06.791 ERROR [org.xcontest.XCTrack.g0$b:c()28] USBConnection
	EXCEPTION: java.io.IOException: Expected 0xee bytes, but get 0xa8 [init #6]
java.io.IOException: Expected 0xee bytes, but get 0xa8 [init #6]
	at org.xcontest.XCTrack.h0.a(USBConnectionCH340.java:8)
	at org.xcontest.XCTrack.h0.d(USBConnectionCH340.java:6)
	at org.xcontest.XCTrack.g0$b.b(USBConnection.java:14)
	at org.xcontest.XCTrack.g0$b.c(USBConnection.java:10)
	at org.xcontest.XCTrack.g0$b.run(USBConnection.java:1)
	at java.lang.Thread.run(Thread.java:920)
```

Because I'm new to arduino I don't have any clue if its a matter on XCTrack's or the programs side. 
Thanks for your help.

dvarrel, 2022-10-03:
Hi,
i open an issue in xctrack, because if it works with an app, i should work with xctrack, and the CH340G driver is supported

dvarrel, 2022-10-05:
i test with xctrack 0.9.8.3 :
```
2022-10-05 05:59:36.827 DEBUG [USBConnection] USB Found Qinheng CH340
2022-10-05 05:59:36.827 INFO [USBConnection] Requesting permission for USB device
2022-10-05 05:59:37.508 DEBUG [Info] adding loc: SensorsLocation{isReal=true, isValid=true, time=1664949577000, rolloverOffset=RolloverOffset{builtin=0, configured=0}, coord=LonLat{lon=5.6853492, lat=46.18162723333333}, altGps=780.7, speedGps=0.0, bearingGps=0.0, headingCompass=null, airspeedSensor=null, pressure=null, speedComputed=0.014818837062232604, bearingComputed=239.98402851215545, headingComputed=0.0, airspeedComputed=0.0}
2022-10-05 05:59:38.510 DEBUG [Info] adding loc: SensorsLocation{isReal=true, isValid=true, time=1664949578000, rolloverOffset=RolloverOffset{builtin=0, configured=0}, coord=LonLat{lon=5.685349316666667, lat=46.18162718333333}, altGps=780.6, speedGps=0.0, bearingGps=0.0, headingCompass=null, airspeedSensor=null, pressure=null, speedComputed=0.010563476220340292, bearingComputed=121.7569993163436, headingComputed=0.0, airspeedComputed=0.0}
2022-10-05 05:59:39.505 DEBUG [Info] adding loc: SensorsLocation{isReal=true, isValid=true, time=1664949579000, rolloverOffset=RolloverOffset{builtin=0, configured=0}, coord=LonLat{lon=5.685349266666667, lat=46.18162718333333}, altGps=780.6, speedGps=0.0, bearingGps=0.0, headingCompass=null, airspeedSensor=null, pressure=null, speedComputed=0.003849427021353673, bearingComputed=270.0, headingComputed=0.0, airspeedComputed=0.0}
2022-10-05 05:59:40.069 INFO [USBConnection] Permission granted for USB device
2022-10-05 05:59:40.071 INFO [USBConnection] USB Connecting to serial device
2022-10-05 05:59:40.074 INFO [USBConnection] Connected to usb device.
2022-10-05 05:59:40.076 INFO [USBConnection] USB - Found CH340 interface: 0
2022-10-05 05:59:40.077 INFO [org.xcontest.XCTrack.g0$b:a()21] USB Listing endpoints: 3
2022-10-05 05:59:40.078 DEBUG [org.xcontest.XCTrack.g0$b:a()24] USB endpoint type: 2, direction: 80
2022-10-05 05:59:40.080 DEBUG [org.xcontest.XCTrack.g0$b:a()24] USB endpoint type: 2, direction: 0
2022-10-05 05:59:40.081 DEBUG [org.xcontest.XCTrack.g0$b:a()24] USB endpoint type: 3, direction: 80
2022-10-05 05:59:40.195 DEBUG [USBConnection] USB: line:>���	��_$LK8EX1,93651,0,9999,20,999,*20<
2022-10-05 05:59:40.308 DEBUG [USBConnection] USB: line:>$LK8EX1,93650,0,9999,20,999,*21<
2022-10-05 05:59:40.421 DEBUG [USBConnection] USB: line:>$LK8EX1,93651,0,9999,20,999,*20<
2022-10-05 05:59:40.513 DEBUG [Info] adding loc: SensorsLocation{isReal=true, isValid=true, time=1664949580000, rolloverOffset=RolloverOffset{builtin=0, configured=0}, coord=LonLat{lon=5.685349366666666, lat=46.18162715}, altGps=780.7, speedGps=0.0, bearingGps=0.0, headingCompass=null, airspeedSensor=null, pressure=93650.02999999933, speedComputed=0.008544617158400363, bearingComputed=115.70779164423159, headingComputed=0.0, airspeedComputed=0.0}
2022-10-05 05:59:40.532 DEBUG [USBConnection] USB: line:>$LK8EX1,93652,0,9999,20,999,*23<
2022-10-05 05:59:40.646 DEBUG [USBConnection] USB: line:>$LK8EX1,93651,0,9999,20,999,*20<
2022-10-05 05:59:40.757 DEBUG [USBConnection] USB: line:>$LK8EX1,93652,0,9999,20,999,*23<
2022-10-05 05:59:40.871 DEBUG [USBConnection] USB: line:>$LK8EX1,93651,0,9999,20,999,*20<
2022-10-05 05:59:40.983 DEBUG [USBConnection] USB: line:>$LK8EX1,93651,0,9999,20,999,*20<
2022-10-05 05:59:41.096 DEBUG [USBConnection] USB: line:>$LK8EX1,93652,0,9999,20,999,*23<
2022-10-05 05:59:41.207 DEBUG [USBConnection] USB: line:>$LK8EX1,93651,0,9999,20,999,*20<
```
ok for me

closed 2022-10-05 by dvarrel